ros/solidworks_urdf_exporter #63: Compile problems when the $(SolutionDir) has spaces in it.

bakerhillpins, 2019-05-22:
Clone the repo into a directory that has spaces in it. Build the SW2URDF project:
`1>  Processing -File 'C:\Users\xxx\Advanced' failed because the file does not have a '.ps1' extension. Specify a valid Windows PowerShell script file name, and then try again.`

To fix the compile error wrap the paths in quotes:
`<PreBuildEvent>"C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe" -ExecutionPolicy Bypass -File "$(SolutionDir)scripts\UpdateVersionInfo.ps1" --filename "$(ProjectDir)Versioning\VersionInfo.cs"</PreBuildEvent>`

For users of VS2017 This can be done directly in VS by unloading the project, right clicking on the project again and editing to insert the above line, then save and reload the project.

brawner, 2019-05-24:
Thanks for the bug report. Spent too much time following linux conventions. Could you check the above referenced pull request?

brawner, 2019-05-25:
Adressed by #66

closed 2019-05-25 by brawner