ros/solidworks_urdf_exporter #64: "Next" button is missing in "Preview and Export" window.

grozail, 2019-05-23:
Using Solidworks 2017, I tried to export model to URDF.
After filling all the required information, I used Preview and Export button.
All the needed values were represented correctly.
But there was no "Next" button so that I could configure dynamic properties of the model.

brawner, 2019-05-24:
This is indeed strange, and not something I have heard being experienced from the few people have used it before. Did you download the installer or try building from source?

What version of .Net are you using? Can you download and install 4.7.2 from https://dotnet.microsoft.com/download/dotnet-framework

rhenandias, 2019-05-24:
Same here, Solidworks 2016 with .Net 4.8 installed

grozail, 2019-05-25:
@brawner pre-built installer v1.5

brawner, 2019-05-25:
What version of Windows is everyone using? 

Would someone be willing to upload a log file? There should be a directory in your home folder labeled sw2urdf_logs. Compress it and reply with a link.

brawner, 2019-05-25:
Actually, could you use the following installer. It adds a bit more information in the log file about the window controls.

https://github.com/ros/solidworks_urdf_exporter/blob/missing_next/INSTALL/Output/sw2urdfSetup.exe

brawner, 2019-05-25:
Sorry, to clarify. I would need to install the new version and proceed to the export window like you did before. Once there, you can click cancel and send me the log file I mentioned.

grozail, 2019-05-25:
> Sorry, to clarify. I would need to install the new version and proceed to the export window like you did before. Once there, you can click cancel and send me the log file I mentioned.

I will try this during next week.
About OS. It was Windows 8.1
Thanks in advance for your work.

rhenandias, 2019-05-25:
> Actually, could you use the following installer. It adds a bit more information in the log file about the window controls.
> 
> https://github.com/ros/solidworks_urdf_exporter/blob/missing_next/INSTALL/Output/sw2urdfSetup.exe

Hey, this version actually worked just fine, I don't get it ... anyway, the log folder still has the yesterday log, here is the link, if it's anyhow useful

Windows 10, Solidworks 2016, .Net 4.8 [Log Folder](https://www.dropbox.com/s/z1wegj4lxsztdq7/sw2urdf_logs.rar?dl=0)

grozail, 2019-05-31:
The "Next" button appeared when installed the version you linked, but when the "Export with Meshes" button was pressed system exporter stopped to respond.
[sw2urdf_logs.zip](https://github.com/ros/solidworks_urdf_exporter/files/3242273/sw2urdf_logs.zip)

jacknlliu, 2019-07-28:
@grozail I met the same issue. Could you put the binary installer on the github release page, not in the source code? And could you provide the different binary installer for the different versions of this repo?

brawner, 2019-08-01:
Addressed in PR #74 I believe. Please reopen this issue if was reverted. Installer coming soon

closed 2019-08-01 by brawner

brawner, 2019-08-02:
Commenting to say installer with this fix is now on releases page.
https://github.com/ros/solidworks_urdf_exporter/releases/tag/1.5.1