ros/solidworks_urdf_exporter #65: [Error] Object reference not set to an instance of an object

mutr, 2019-05-24:
`There was a problem with the configuration property manager:  "Object reference not set to an instance of an object."`

Plugin version: 1.5 (from wiki).
Solidworks version checked: 
2019 SP3.0 Premium
2016 SP3.0 Premium
Both have the same issue.

[sw2urdf.log](https://github.com/ros/solidworks_urdf_exporter/files/3215674/sw2urdf.log)

mutr, 2019-05-24:
[image]

How to reproduce:
1. Download [urdf_test16.zip](http://www.mediafire.com/file/wdf2blu1qothe1h/urdf_test16.zip/file)
2. Open urdf_test.SLDASM
3. Select "Export as URDF"
4. Click "Preview and Export"

brawner, 2019-05-24:
Thanks for the bug report. I can reproduce your issue. While the error message is very unhelpful and needs to be fixed for this case, the underlying issue is that bone_31 in your configuration doesn't have a part associated with it. Alternatively, if you'd like to leave that empty, change the joint type from "Detect automatically" to "fixed"

closed 2019-08-01 by brawner