ros/solidworks_urdf_exporter #67: Does not start

dbc, 2019-06-14:
See this Q on rosanswers: https://answers.ros.org/question/325713/sw2urdf-v15-wont-activate-in-sw2018/

In addition to my report, another user reports that sw2urdf seems to install correctly but does not start.  I'm not really sure where to start debugging this, and I am not set up for Windows development.  Is there something I can do to start debugging this without having to spin up a Windows development environment?

brawner, 2019-06-14:
The exporter creates a log file in sw_logs in your home directory. Could you please attach an archive of that directory? It might shed light on the issue.

dbc, 2019-06-14:
The other user on rosanswers just posted that installing with full admin privileges cleared the issue for him. As soon as I have access to the Solidworks machine again I will first try re-install making sure to use full admin, and if that fails will pull the log.

brawner, 2019-06-14:
Oh good to know. I'll work on creating a more transparent installer in that regard.

dbc, 2019-06-14:
OK, have an update.  My M.E. workmate is working remotely today, so I don't have the SW machine with me. they tried this and report the following:

- The user account has "Administrator" privileges, and when installed from there, the plug-in does not start.
- After re-installing when logged in as "IT Administrator", it runs, but fails.
- As the normal user (with "Administrator" priv) it still does not run, even after being installed by "IT Administrator".
- When run as IT Administrator, it fails with the error dialog: "Object reference not set to an instance of an object." and no files have been saved to the location described by http://wiki.ros.org/sw_urdf_exporter/Tutorials/Export%20a%20Part

They neglected to send me a log, but I will ask for one and attach it when I get it.

dbc, 2019-06-18:
We are on service pack 4 of SW2018

dbc, 2019-06-21:
Past the permission issue but stuck on the object reference crash. It appears to me that the message:  "Object reference not set to in instance of an object." is a generic Windows scripting error message.  (I may be wrong here, as I do little Windows development.)  In any case, I am thinking that this message could come from a number of causes that all end up in the "null reference" category.  So... maybe something changed in recent service packs of Solidworks such that the sw2urdf script doesn't fetch something any more?  Is there some setting missing in the SW model that we may have overlooked?

brawner, 2019-08-01:
Looks like the install issue was resolved and the object reference crash has been resolved in #72.

closed 2019-08-01 by brawner

mbait, 2019-10-17:
I would like to re-open this issue because I experience exactly the same behavior on SW2018 SP4/SP5 - the plug-in installs just fine, but would never run thereafter. I tried to run the plug-in installer and SW on behalf on regular user as well as "Run as Administrator" - still nothing. When I check the checbox in the "Add-ins" menu in SW and hit "OK", nothing appears under "Menu -> File", and the checkbox becomes unchecked when I open "Add-ins" after that.

mbait, 2019-10-17:
And the log files are pretty much empty:
```
2019-10-17 10:27:37,691 INFO  Logger.cs: 70 - 
--------------------------------------------------------------------------------
2019-10-17 10:27:37,723 INFO  Logger.cs: 71 - Logging commencing for SW2URDF exporter
2019-10-17 10:27:37,723 INFO  Logger.cs: 73 - Commit version 1.5.1-0-g916b5db
2019-10-17 10:27:37,723 INFO  Logger.cs: 74 - Build version 1.5.7152.31018
```

blubbi321, 2020-07-27:
Also have the issue on Solidworks Premium 2020 SP3.0

I have a few logs that look like @mbait 's above and some that have the two additional lines as below

```
2020-07-27 12:00:16,420 INFO  Logger.cs: 70 - 
--------------------------------------------------------------------------------
2020-07-27 12:00:16,452 INFO  Logger.cs: 71 - Logging commencing for SW2URDF exporter
2020-07-27 12:00:16,453 INFO  Logger.cs: 73 - Commit version 1.5.1-0-g916b5db
2020-07-27 12:00:16,453 INFO  Logger.cs: 74 - Build version 1.5.7152.31018
2020-07-27 12:00:16,454 INFO  SwAddin.cs: 109 - Registering SOFTWARE\SolidWorks\Addins\{65c9fc17-6a74-45a3-8f84-55185900275d}
2020-07-27 12:00:16,455 INFO  SwAddin.cs: 117 - Registering Software\SolidWorks\AddInsStartup\{65c9fc17-6a74-45a3-8f84-55185900275d}

```

blubbi321, 2020-07-27:
Checked by going through the development instructions and running the plugin in debug mode from visual studio 2017.

It looks as if the "ConnectToSW()" method is never called on my machine (which explains the log output). 

Any help is much appreciated ..

blubbi321, 2020-07-28:
I figured that I can download older Solidworks versions through the customer portal with my license. Have the exporter working now, Solidworks Premium 2019 SP3.0 (<= guessed this one according to release date)

Log now looks like this:
```
2020-07-28 09:52:20,839 INFO  Logger.cs: 70 - 
--------------------------------------------------------------------------------
2020-07-28 09:52:20,858 INFO  Logger.cs: 71 - Logging commencing for SW2URDF exporter
2020-07-28 09:52:20,859 INFO  Logger.cs: 73 - Commit version 1.5.1-0-g916b5db
2020-07-28 09:52:20,859 INFO  Logger.cs: 74 - Build version 1.5.7152.31018
2020-07-28 09:52:20,866 INFO  SwAddin.cs: 250 - Adding Assembly export to file menu
2020-07-28 09:52:20,866 INFO  SwAddin.cs: 253 - Adding Part export to file menu
2020-07-28 09:52:20,869 INFO  SwAddin.cs: 218 - Connecting plugin to SolidWorks
2020-07-28 09:56:06,961 INFO  SwAddin.cs: 328 - Part export called
[..]
```