ros/solidworks_urdf_exporter #68: Object reference not set to an instance of an object

prasuchit, 2019-06-15:
Hi, I'm trying to use this package to export a few parts from solidworks. I followed https://wiki.ros.org/sw_urdf_exporter instructions first and then found this repo. I installed it as per instructions and while I tried to execute it on Visual Studio, it failed. The errors can be seen in the attached screenshot.
[image]

Then, I allowed it to run the last working build and it opened solidworks. This is what I have when I try to export as urdf:
[image]

I keep getting the error "Object reference not set to an instance of an object". The only difference is, after installing this package, I'm able to see the line of code where the error is occuring:
[image]
But I don't understand what is causing this error. All the fields exist for the model and they have been given valid values and defined properly.

What am I missing? I have attached the model below in case you want to reproduce the error with it.
[Frame.zip](https://github.com/ros/solidworks_urdf_exporter/files/3292660/Frame.zip)

 Kindly help, I'm on a clock for a project. Thanks in advance.

dbc, 2019-06-18:
I wonder if this is the same problem that we are seeing.  What version and service pack of Solidworks are you on?  We will check our service pack and append to our issue.  It seems like the inertias are not getting passed to the plugin.

prasuchit, 2019-06-18:
Hi @dbc,
I'm using Solidworks 2017 x64 edition, SP 5.0.
Kindly let me know if you need anything further. Thanks.

dbc, 2019-06-21:
We are on different version of SW, but recent service pack for our version.  I wonder if recent service packs of SW for all versions have changed the API such that sw2urdf can not longer find something?  Just speculating.  But there are 3 open issues that all hit the error "Object reference not set to an instance of an object", which I gather is a generic Windows scripting error about a null reference. (I don't do Windows development, so might be off base here...)  

I would bet that there are multiple ways to trigger the "Object reference not set..." message, but I don't know if there are ways to get a better traceback into the sw2urdf script.  It might require adding a bunch of null reference checks within the script itself to narrow down the issues people are seeing.

prasuchit, 2019-06-21:
Hi @dbc, could you tell me which version, service pack and windows version you are working on? Or which version this plugin works seamlessly in?

Alternatively, if you/anyone managing this package could give me a way to fix/get-around this problem, that would be great! As I said, I'm on a clock for this project and I need to show results asap. Thanks.

dbc, 2019-06-21:
We are on service pack 4 of SW2018. Exporting a single part, not an assembly. Windows 10. 

Sorry I can't help with pointing you at a version that works... I am not a maintainer, just trying to use the package for the first time. 

If I were to guess, I would say try rolling back to earlier service packs of SW.  I know that is a bunch of work, and it might not work at all and just be a waste of your time.  I am just saying that because it smells like the kind of bug that comes up because "something moved" between service packs.

You say you have everything defined... but I assume you saw the other "object reference" issue that is open? That was resolved by tweaking a joint definition.  It might be worth double-checking all of your  definitions in SW before going through a bunch of hassle with service pack roll-backs.  Anything left unspecified is a possible trigger.

Extent421, 2019-07-02:
This comes down to line 147 of ExportHelperExtension.cs,  AsRowMajorArray() is returning null on the matrix object, only AsColumnMajorArray() returns an actual array for some reason, though the comment above specifically calls out the swap, but I don't really know why.

prasuchit, 2019-07-02:
@dbc, @Extent421, since you have the working sw2urdf plugin, if I send you my model, could you export it as urdf and send it to me? I'm running out of time and I badly need to get this working and writing a urdf from scratch is going to take a lot of time that I don't have. Kindly help me out with this. Thanks in advance.

Extent421, 2019-07-03:
sorry I don't have time to help with your assignment.  I've only isolated the null ref, I don't actually have a fix.  You might try and find an old version of the plugin perhaps, it looks like this may have only changed with the math library change recently.

brawner, 2019-08-01:
Addressed by #72

closed 2019-08-01 by brawner