ros/solidworks_urdf_exporter #69: Problem with Moveit Setup Assistant ROS

mictas, 2019-06-21:
Hello,
I'm new about Solidworks_urdf_exporter and I'm sorry if my question can seem trivial.
For an our project I'm using ROS Kinetic on Ubuntu 16.04 to manage a robotic arm.
We used the last release of Solidworks_urdf_exporter to get the urdf file of our robot project.
When I import the .urdf file in "MoveIt Setup Assistant" it works correctly, but when I try to set the robot poses in "Define Robot Poses Section" the joints are blocked and I can't move the robot.
I didn't found this type of error in MoveIt Setup Assistant when we used the previous version of "Solidworks_urdf_exporter" to get the urdf file.
Where can I download the previous version of Solidworks_urdf_exporter to check if it is a problem of my project?
Thank you.

brawner, 2019-06-21:
Old versions are still available on BitBucket (for now). I'll probably close that repo eventually.

https://bitbucket.org/brawner/sw2urdf/src/default/INSTALL/Output/

mictas, 2019-06-24:
Thank you for your response. Using the current version of sw2urdf I found in the .urdf file that the lines "<safety_controllers k_velocity="0"/>" are added automatically for each joint. These lines weren't added using previous version of sw2urdf. Removing them from the .urdf file, MoveIt Setup Assistant now works fine.

closed 2019-06-24 by mictas