ros/solidworks_urdf_exporter #70: URDF file exported from SW2018 has a problem in Moveit!

13651924886, 2019-06-25:


brawner, 2019-08-01:
If this works fine with the joint state publisher, then I'm not sure what the MoveIt issue is. Maybe @davetcoleman has some ideas

cesic, 2019-08-09:
I have the same question

davetcoleman, 2019-08-13:
I'm not sure what would cause that except if your joint value range was zero, or maybe really really small.. ?

Seems like a question for the https://github.com/ros-planning/moveit repo instead of here.

Btw the logic for those sliders is here:
https://github.com/ros-planning/moveit/blob/master/moveit_setup_assistant/src/widgets/robot_poses_widget.cpp#L477

Feel free to debug yourself and suggest a PR ;-)

caioaamaral, 2020-01-23:
does your URDF have transmission tags?

brawner, 2020-06-12:
Closing due to age. It also doesn't seem like the issue is with this code.

closed 2020-06-12 by brawner