ros/solidworks_urdf_exporter #73: Exporting mesh with wrong origin

0Nel, 2019-07-31:
I planned on using the new version of the urdf exporter to generate a valid urdf of a huge solidworks assembly. Beforehand I wanted to verify the workflow with just the base_link and one attached link. I defined each part to be located in it's origin, gave the revolute joint an appropriate axis and an own origin as well and upon exporting I chose these predefined origins. While the location of the joints looks right the location of the visual/collision meshes is off. I checked the exported meshes in meshlab and was able to visualize that the meshes origin is indeed wrong and not at the intended location (It seems to be off by exactly the vector that lies between the base_links origin and the joint origin).
After trying every approach i could think off I ended up reinstalling SolidWorks and installing the old (1.2 I believe) version of the urdf_exporter, which works like a charme.

(Sadly I cannot share my CAD files and since I actually had to reinstall SolidWorks and the exporter I am not willing to set up a test case, but I hope that this issue is easily reproducable.)

Best luck in finding a quick fix!

brawner, 2019-08-01:
This is a known bug in SolidWorks 2018 and you need to upgrade to Service Pack 5 or SW 2019.

closed 2019-08-01 by brawner

DentOpt, 2021-01-28:
In the 2020 editions the meshes are exported in respect to the origin even when the appropriate link reference frames are selected in the export procedure.  Is there a workaround?