ros/solidworks_urdf_exporter #92: Computed mass is more than actual model property

saikishor, 2019-10-10:
We are trying to export the urdf from the solidworks, and we found that the plugin is somehow adding more mass to the exported URDF (around 20% higher). Is there a reason for this?. How can we make the plugin use the actual mass from the solidworks model?

brawner, 2019-10-11:
It uses the mass properties in the components of Solidworks. It shouldn't disagree with the mass properties unless you have overridden them. I looked into it awhile ago and could't figure out a way to get the overridden mass properties though. Is that possibly what's happening?

Otherwise, do you have a solidworks assembly you might be able to attach or send me so I can see your issue?

saikishor, 2019-10-18:
Yes we are overriding the mass properties because we customize the materials in the later stage, is it not possible to recover the overridden mass?

topherbuckley, 2020-06-09:
I think this should be closed and consolidated to [Issue 93](https://github.com/ros/solidworks_urdf_exporter/issues/93)

brawner, 2020-06-09:
I agree, this sounds like the same problem. Moving discussion over there.

closed 2020-06-09 by brawner