ros/solidworks_urdf_exporter #95: Problems when V-REP importing URDF

Munrel, 2019-11-22:
Hi, I have a problem to ask for help. When V-REP imports the urdf file export by SW2018, it has errors: "ERROR: file couldn't be opened." and "ERROR: there is no robot in the file." Could you give me some ideals? Thank you.

brawner, 2019-11-22:
Are you able to launch the rviz example file? If you can attach the urdf package and solidworks assembly, it might help illustrate the issue.

Munrel, 2019-11-25:
I will have a try, thank you for your help very much~

brawner, 2020-06-12:
Closing this issue, due to age. Let me know if it needs to be reopened.

closed 2020-06-12 by brawner

charlotte-sy, 2020-06-17:
Have you guys got any updates? Same issue here. 

I have tried in rvis as well but there is no model is shown with no error