ros/solidworks_urdf_exporter #96: [ImportError: cannot import name 'TransformException']

ziyadAbouelenin, 2020-01-22:
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/gazebo_ros/spawn_model", line 32, in <module>
    import tf.transformations as tft
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
ImportError: cannot import name 'TransformException'
**[spawn_model-5] process has died [pid 24326, exit code 1, cmd /opt/ros/kinetic/lib/gazebo_ros/spawn_model -file /home/ziyad/catkin_ws/src/cf_gazebo_model/urdf/cf_gazebo_model.urdf -urdf -model cf_gazebo_model __name:=spawn_model __log:=/home/ziyad/.ros/log/5274d232-3d60-11ea-a05a-1cbfcea1244c/spawn_model-5.log].**

------------------------------------------------

I did the steps exactly but can not get the model to appear in gazebo. I ,oved the output package from sw2urdf to my catkin work space, build it with no errors. However when trying to launch the gazebo.launch file, I get the above error. 

Thank you in advance for your support.

brawner, 2020-01-22:
This looks like an issue with your ROS setup or code your running on ROS and is unrelated to the URDF generated from this Solidworks add-in. I would suggest asking a question on ros answers (answers.ros.org) or filing an issue over at tf2_ros (https://github.com/ros/geometry2/issues).

closed 2020-01-22 by brawner

ziyadAbouelenin, 2020-01-22:
Thank you Brawner, it was a problem with my ROS the installation. Anyways, I designed the CAD file using solidworks 2019-2020  and used your package. The stl file looks great in gazebo thanks to you and correctly identifies the joints. I will recommend the guideline to anyone using your great package: [https://blogs.solidworks.com/teacher/wp-content/uploads/sites/3/WPI-Robotics-SolidWorks-to-Gazebo.pdf](url) Again, Thank you very much.