ros/solidworks_urdf_exporter #97: Cannot export a (very) simple 2 wheels (due to simplicity) car from Solidworks to URDF for Gazebo.

tzoutzoukos, 2020-01-30:
Dear sir,
I am trying to export a very simple car from Solidworks to URDF so, I can import it into Gazebo.
The car (which should have 4 wheels) has only 2 wheels and that for simplicity.
Although the less than minimum configuration of the model, I cannot seem to find the way to export to URDF.
I have tried every combination of defining Axis and Coordinate systems for the location of both the chassis and the wheels (both in the Assembly level as well as the part level).
I have checked the Assembly as far as the definition of mates is correct and fully defined so, I really don't know what else to check.
The model consists of 
A central 200X100X10mm plate
Four 10mm diaX10mm height cylinder to play the role of the wheel axle
Four 50mm dia X 10mm thickness cylindrical pieces playing the role of the wheel, having a hole of 10mm at the center.
The four wheels have been reduced to 2 (for the sake of ultimate simplicity)

Results: 
Left wheel in place
Right wheel bouncing around near cylindrical axle off place.

Any ideas on how to proceed would be most appreciated.

BR

brawner, 2020-01-30:
This might be better asked on answers.ros.org so everyone can benefit from the response. If you post this question, with a copy of your SolidWorks model, I can try to deduce your issue. Please tag the answer with `sw_urdf_exporter` and post a link on this bug report.

tzoutzoukos, 2020-01-30:
Have already asked answers.ros.org.
Please check the link: 
https://answers.ros.org/question/342810/solidworks-to-urdf-export-incorrect-wheel-location/

I tagged with requested sw_urdf_exporter.

I cannot upload anything there due to low rep.

BR

brawner, 2020-01-30:
Are you able to provide a download link on this issue? Without the model,
it's nearly impossible to guess as to what the issue is.

tzoutzoukos, 2020-01-30:
Of course I can here you are:
https://ufile.io/940vjvfb

This is the Solidworks model.

tzoutzoukos, 2020-01-31:
Have you managed to figure what is wrong?

brawner, 2020-01-31:
I don't have time until early next week to look at this. Sorry

brawner, 2020-02-04:
As I mentioned on the answers post, this seems like it is related to the STL export bug in SolidWorks 2018. I had no problems exporting this SW model and loading it into rviz, and everyone looks as expected. Recommend upgrading to SW 2018 SP 5 or newer.

closed 2020-02-04 by brawner

tzoutzoukos, 2020-02-05:
ok, I have confirmed your suggestion. It was indeed an issue with Solidworks 2018 SP<5.0. 

Upgraded to SW2019 SP5.0 and problem solved!
I didn't verify if it was really the STL exporter causing the issue, but I take your word for it.

Thank you very much for your support.