ros/solidworks_urdf_exporter #101: Masses and moments of inertia not using hidden parts/assemblies

topherbuckley, 2020-06-09:
I followed [the tutorial](http://wiki.ros.org/sw_urdf_exporter/Tutorials/Organizing%20SolidWorks%20Assembly%20for%20URDF%20Exporter) regarding how to set up an assembly for a simpler mesh and in the export process I noticed the mass of the links seems to correspond to that of air (material set for skin parts) and not the mass of the hidden (but not suppressed) parts and assemblies. Is this a known issue, and if so how can we work around this to have both simplified meshes and accurate mass and moments of inertia?

I am only including the skin parts in the initial Link Components" GUI, and two child links attached to the base link. All these links correspond to the skin parts. Is this correct or should I be selecting ALL my normal parts in this part of the export process?

brawner, 2020-06-09:
Please ask this question over at answers.ros.org and tag it sw_urdf_exporter. Feel free to add a link to the question in this issue.

closed 2020-06-09 by brawner

topherbuckley, 2020-06-10:
Done. [It](https://answers.ros.org/question/354487/masses-and-moments-of-inertia-do-not-match-solidworks-mass-properties-when-using-solidworks-urdf-exporter/) is awaiting moderation.