ros/solidworks_urdf_exporter #102: ERROR: file couldn't be opened. ERROR: there is no robot in the file.

charlotte-sy, 2020-06-17:
I have created URDF and mesh files. Even when I open urdf file, there is code which seems okay.

However, when I open the urdf file in V-Rep, I got ERROR: file couldn't be opened. ERROR: there is no robot in the file.

I have tried to re-install urdf installer using visual studio as well. Still same issue..... 

Is anyone who faced the same or know how to solve this issue?

brawner, 2020-06-17:
Are you able to make available a copy of the package that was exported?

charlotte-sy, 2020-06-18:
Even a simple box has same issue.

brawner, 2020-06-18:
This is not a bug that many have experienced, at least when using ROS. I don't test the output with V-Rep.

Please provide a representative SolidWorks model that reproduces this issue so I can better understand it.

johncgh22, 2020-08-14:
Hi @brawner and @charlotte-sy 

I´m having the same issues with the URDF Exporter in CoppeliaSim (aka V-REP)
I'm working with SolidWorks 2019 SP5.
I'll attach my assembly if helps. It's on Onshape.

[image]

[https://cad.onshape.com/documents/ed9c53435ea47fa8b40406fd/w/c5cb31884dcfc2499f9b8f26/e/d586ffe1e96e33eaa2029bd5](url)

OrionHeart, 2021-04-04:
Also getting the same error, did you guys figured out how to use it on V-REP?

OrionHeart, 2021-04-04:
I figured out that it was not working because the path to the URDF contained "Investigación" (as I am an spanish-native speaker) which had an illegal character, I removed it and it worked.

johncgh22, 2021-04-08:
Hi @OrionHeart 
I can confirm that having ilegal characters in the path of the URDF file makes this error. So for CoppeliaSim/V-REP, just make sure your path doesn´t have any ilegal characters and you should have your robot exported.

In @charlotte-sy case, I recommend that you specify STL files for the visuals and collisions for each part of your robot inside your URDF file, something like this:
`<visual>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://serial3_robot_v3/meshes/visual/base_link.STL" />
      </geometry>
      <material
        name="">
        <color
          rgba="0.75294 0.75294 0.75294 1" />
      </material>
    </visual>
    <collision>
      <origin
        xyz="0 0 0"
        rpy="0 0 0" />
      <geometry>
        <mesh
          filename="package://serial3_robot_v3/meshes/collision/base_link.STL" />
      </geometry>
    </collision>`

You can make these STL files in SolidWorks by using Save as, then choose STL, then click on Options. For Visuals, select **High Quality Resolution**, for Collisions, select **Fast Resolution**. Also, select the corresponding output coordinate system for each of your robot parts to make sure that they can be located, this option is at the bottom of the window. 
Then, put the corresponding STL files inside their specific folders inisde the meshes folder. With all these files, you can use the display.launch or gazebo.launch to view your robot.

brawner, 2021-11-21:
Thanks for the suggested fix @johncgh22. Seeing as this is not an issue with the exporter, I'll close this for now.

closed 2021-11-21 by brawner