ros/solidworks_urdf_exporter #104: "Object reference not set to an instance of an object"

lpay0001, 2020-09-21:
Hi there,

How can I fix this issue? I have noticed others have written similar problems however could not see a solution.

[image]

I would greatly appreciate your help.

brawner, 2020-09-21:
It's tough to say because it's a generic error. Does the log message indicated in the error message say anything, could you post the last page or so?

If you're familiar with building from source, you can run in debug mode.

I suspect your design has changed somehow that the tool is not able to handle. Otherwise, you can delete your configuration and start again (unfortunately)

lpay0001, 2020-09-21:
In summary, the last few lines are:

2020-09-21 15:52:43,774 INFO  Common.cs: 230 - Loaded 2 components for link Link_6
2020-09-21 15:52:43,790 INFO  SwAddin.cs: 321 - Showing property manager
2020-09-21 15:53:17,546 INFO  ExportPropertyManager.cs: 456 - Configuration saved
2020-09-21 15:53:17,547 INFO  Serialization.cs: 275 - URDF Configuration found.....................
.................</SWComponentPIDs><SWMainComponentPID i:nil="true"/><Visual z:Ref="i16"/><isFixedFrame>false</isFixedFrame><isIncomplete>false</isIncomplete></Link>
2020-09-21 15:53:17,591 INFO  ExportPropertyManager.cs: 1184 - AfterClose called. This method no longer throws an Exception. It just silently does nothing. Ok, except for this logging message.

I'm unfamiliar with building from the source unfortunately. I will have a go at deleting the configuration. 
If there's anything abnormal you can see in the log, please let me know! Thank you for your time.

lpay0001, 2020-09-27:
I have resolved the error by specifying all joint types, axes and coordinate systems rather than computing automatically with the software. Thank you for your time.

edgarcamilocamacho, 2020-10-26:
@lpay0001 same here, same error and solved just defining all joint types, axes and coordinate systems.

ben-mcdonnell-hacksmith, 2022-01-24:
I am getting the same error on export, but the log file I have is referencing files that aren't on my computer. Is this expected behavior?

`
2022-01-24 11:38:35,608 ERROR ExportPropertyManager.cs: 399 - Exception caught handling button press 17
System.NullReferenceException: Object reference not set to an instance of an object.
   at SW2URDF.URDFExport.ExportHelper.AddLimits(Joint Joint, List`1 limitMates, Component2 parentComponent, Component2 childComponent) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 1259
   at SW2URDF.URDFExport.ExportHelper.EstimateGlobalJointFromComponents(Link parent, Link child) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 892
   at SW2URDF.URDFExport.ExportHelper.CreateJoint(Link parent, Link child) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 425
   at SW2URDF.URDFExport.ExportHelper.CreateLinkFromComponents(Link parent, LinkNode node) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 348
   at SW2URDF.URDFExport.ExportHelper.CreateLink(LinkNode node, Int32 count) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 215
   at SW2URDF.URDFExport.ExportHelper.CreateLink(LinkNode node, Int32 count) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 227
   at SW2URDF.URDFExport.ExportHelper.CreateRobotFromTreeView(LinkNode baseNode) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportHelperExtension.cs:line 169
   at SW2URDF.URDFExport.ExportPropertyManager.ExportButtonPress() in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportPropertyManager.cs:line 239
   at SW2URDF.URDFExport.ExportPropertyManager.OnButtonPress(Int32 Id) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportPropertyManager.cs:line 381
   at SW2URDF.URDFExport.ExportPropertyManager.SolidWorks.Interop.swpublished.IPropertyManagerPage2Handler9.OnButtonPress(Int32 Id) in C:\Users\Stephen Brawner\workspace\solidworks_urdf_exporter\SW2URDF\URDFExport\ExportPropertyManager.cs:line 399
2022-01-24 11:38:35,653 INFO  ExportPropertyManager.cs: 1142 - AfterClose called. This method no longer throws an Exception. It just silently does nothing. Ok, except for this logging message
` 

Thanks!