ros/solidworks_urdf_exporter #105: Unable to export to urdf in solidworks 2020

diwakarshastri, 2020-09-26:
I have a developed an assembly in solidworks 2020 and I want to export it to .urdf. I dont find an export option in solidworks 2020. I request you to fix the issue at the earliest 

thank you

acbuynak, 2020-09-27:
I also have issues using this Add-In with Solidworks 2020. Following the suggested debugging steps in the [README.md](https://github.com/ros/solidworks_urdf_exporter/blob/master/README.md), I was still unable to resolve my issues.

@diwakarshastri Have you attempted these steps yet. If you already have, I'd encourage you to post more details of your build errors for other more knowledge users (not me) to help you.  -Cheers

-----------------
**My Errors**
Using the provided .exe file, I saw a subfolder created in Solidworks, but when was unable to load the program in the "AddIn" options in Solidworks.

Using "Debug" from Visual Studio 2019, I had 0 errors and 7 warnings. The afore mentioned subfolder was not created. Though I do see an option in the Solidworks "AddIn" list (weird). This still does not let me add the program.

User/sw2urdf_logs:
Nothing special to note in the logs.

Summary of Warnings from Visual Studio 2019:
```
Severity	Code	Description	Project	File	Line	Suppression State
Warning		Type library exporter warning processing 'SW2URDF.UI.IMessageBox.Show(#0), SW2URDF'. Warning: Non COM visible value type 'System.Windows.MessageBoxResult' is being referenced either from the type currently being exported or from one of its base types.	SW2URDF			
Warning		Type library exporter warning processing 'SW2URDF.UI.IMessageBox.Show(#0), SW2URDF'. Warning: Non COM visible value type 'System.Windows.MessageBoxResult' is being referenced either from the type currently being exported or from one of its base types.	SW2URDF			
Warning		Registering an unsigned assembly with /codebase can cause your assembly to interfere with other applications that may be installed on the same computer. The /codebase switch is intended to be used only with signed assemblies. Please give your assembly a strong name and re-register it.	SW2URDF	C:\Users\acbuynak\GIT DOWNLOAD\exporter\SW2URDF\RegAsm		
Warning	CA1063	Provide an overridable implementation of Dispose(bool) on 'SWTestFixture' or mark the type as sealed. A call to Dispose(false) should only clean up native resources. A call to Dispose(true) should clean up both managed and native resources.	SW2URDF	C:\Users\acbuynak\GIT DOWNLOAD\exporter\SW2URDF\Test\SWTestFixture.cs	10	Active
Warning	CA1816	Change 'SWTestFixture.Dispose()' to call 'GC.SuppressFinalize(object)'. This will prevent derived types that introduce a finalizer from needing to re-implement 'IDisposable' to call it.	SW2URDF	C:\Users\acbuynak\GIT DOWNLOAD\exporter\SW2URDF\Test\SWTestFixture.cs	26	Active
Warning	CA1063	Modify 'SWTestFixture.Dispose()' so that it calls Dispose(true), then calls GC.SuppressFinalize on the current object instance ('this' or 'Me' in Visual Basic), and then returns.	SW2URDF	C:\Users\acbuynak\GIT DOWNLOAD\exporter\SW2URDF\Test\SWTestFixture.cs	26	Active
Warning	CA0507	Post-build Code Analysis (FxCopCmd.exe) has been deprecated in favor of FxCop analyzers, which run during build. Refer to https://aka.ms/fxcopanalyzers to migrate to FxCop analyzers.				Active
```

diwakarshastri, 2020-09-28:
Hi @acbuynak, yes I did try those steps. I got close to 300 errors. 
I did not look into solving those errors.

Kudos

diwakarshastri, 2020-09-29:
Hi @brawner, when can I expect the issue to be resolved?

mis-eu, 2020-09-30:
_issue:_ successful build of SW2URDF v.1.4.4 with Visual Studio **2017**  on Windows 18363 does work in SolidWorks 2020.
Screenshot below, where the Add-in is visible but can not be loaded (even if checked), holds true also for the installer [sw2urdfsetup.exe](https://github.com/ros/solidworks_urdf_exporter/releases/download/1.5.1/sw2urdfSetup.exe)

[image]

notes: 

v.1.5.1 source code **could not** be build with Visual Studio 2017 on Windows 18363 due to` testrunner` 
[image]
[image]

v.14.4 source code **can be built** with Visual Studio 2017 on Windows 18363 with the full download of .NET and SDK, 
as well as granting full R-W-R permissions in Terminal to the source code directory.
[image]

additional debug tips:

if Build is failing, delete any `<HintPaths>` from SW2URDF.csproj
[image]

mis-eu, 2020-09-30:
_question:_

do these Reg keys play a role in API compatibility with particular SW release?

[Run]                                                        
Filename: "{reg:HKLM64\SOFTWARE\Microsoft\.NETFramework,InstallRoot}\v4.0.30319\RegAsm.exe"; Parameters: """{app}\SW2URDF.dll"" ""/codebase"""; StatusMsg: Registering controls ...; Check: IsWin64; Languages:

[Registry]
Root: HKLM64; Subkey: "SOFTWARE\SolidWorks\Addins\65c9fc17-6a74-45a3-8f84-55185900275d";        ValueType: none; ValueName: ""; Flags: dontcreatekey deletekey uninsdeletevalue; Check: IsWin64
Root: HKCU64; Subkey: "Software\SolidWorks\AddInsStartup\65c9fc17-6a74-45a3-8f84-55185900275d"; ValueType: none; ValueName: ""; Flags: dontcreatekey deletekey uninsdeletevalue; Check: IsWin64

[UninstallRun]

Filename: "{reg:HKLM64\SOFTWARE\Microsoft\.NETFramework,InstallRoot}\v4.0.30319\RegAsm.exe"; Parameters:  """{app}\SW2URDF.dll"" ""/unregister"""; StatusMsg: Unregistering controls ...; Check: IsWin64; Languages:

brawner, 2020-10-07:
I was able to reproduce the issue locally. There appears to be an issue with how I catch exceptions for logging as a SolidWorks add-in. Still investigating.

Commenting out the code in this constructor lets the exporter run, but logging exceptions will not work.
https://github.com/ros/solidworks_urdf_exporter/blob/2c515a48b296fa1654dbaffd62ae203a241fc847/SW2URDF/SW/SwAddin.cs#L175-L176

brawner, 2020-10-07:
I believe I've addressed the main cause of the issue with #106, which I'll have to take some time to cleanup and try to address other outstanding issues. In the meantime can someone verify that result with this new installer? It's unsigned, so you'll need Admin privileges on your computer.

https://github.com/ros/solidworks_urdf_exporter/blob/58c2f2a7c55a5084618ef1c3007defb5a2b9ef51/INSTALL/OUTPUT/sw2urdfSetup.exe?raw=true

brawner, 2020-10-09:
Can someone please test the above installer and let me know if it resolves the issue for them?

acbuynak, 2020-10-09:
@brawner I am currently having unrelated OS issues on my system that I have been working to resolve, but will try to test the proposed changes later tonight.

mis-eu, 2020-10-12:
> Can someone please test the above installer and let me know if it resolves the issue for them?

thanks very much for these updates @brawner 

the new **sw2urdfSetup.exe** installer now seems to LOAD as an Add-in, 
but is still not accessible under Tools
[image]

as for Building from updated source code on Branch https://github.com/ros/solidworks_urdf_exporter/tree/compatibility-2020

what would be the proper way of loading/referencing all dependencies in VS2017 -- 

<h3> Assembly error log</h3>
<details>
Severity	Code	Description	Project	File	Line	Suppression State
Warning		The referenced component 'solidworkstools' could not be found.	SW2URDF			
Error	CS8107	Feature 'default literal' is not available in C# 7.0. Please use language version 7.1 or greater.	SW2URDF	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\SW2URDF\URDFExport\ExportHelperExtension.cs	928	Active
Error	CS8107	Feature 'default literal' is not available in C# 7.0. Please use language version 7.1 or greater.	SW2URDF	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\SW2URDF\URDFExport\ExportHelperExtension.cs	1017	Active
Error	CS8107	Feature 'default literal' is not available in C# 7.0. Please use language version 7.1 or greater.	SW2URDF	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\SW2URDF\Utilities\MathOPS.cs	42	Active
Error	CS8107	Feature 'default literal' is not available in C# 7.0. Please use language version 7.1 or greater.	SW2URDF	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\SW2URDF\Utilities\MathOPS.cs	61	Active
Error	CS0246	The type or namespace name 'SolidWorks' could not be found (are you missing a using directive or an assembly reference?)	TestRunner	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\TestRunner\TestRunner.cs	1	Active
Warning		Could not resolve this reference. Could not locate the assembly "SolidWorks.Interop.swconst". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	SW2URDF			
Warning		The referenced component 'SolidWorks.Interop.swconst' could not be found.	SW2URDF			
Warning		Could not resolve this reference. Could not locate the assembly "SolidWorks.Interop.sldworks". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	SW2URDF			
Warning		Could not resolve this reference. Could not locate the assembly "SolidWorks.Interop.swpublished". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	SW2URDF			
Warning		The referenced component 'SolidWorks.Interop.sldworks' could not be found.	SW2URDF			
Warning		The referenced component 'SolidWorks.Interop.swpublished' could not be found.	SW2URDF			
Warning		Could not resolve this reference. Could not locate the assembly "solidworkstools". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	SW2URDF			
Warning	CS8019	Unnecessary using directive.	SW2URDF	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\SW2URDF\Properties\Resources.Designer.cs	12	Active
Warning	IDE0005	Using directive is unnecessary.	SW2URDF	C:\Users\Fiona\Desktop\solidworks_urdf_exporter-compatibility-2020\solidworks_urdf_exporter-compatibility-2020\SW2URDF\Properties\Resources.Designer.cs	12	Active
Warning	MSB3245	Could not resolve this reference. Could not locate the assembly "SolidWorks.Interop.sldworks". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	TestRunner	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets	2110	
Warning	MSB3245	Could not resolve this reference. Could not locate the assembly "SolidWorks.Interop.swconst". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	TestRunner	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets	2110	
Warning	MSB3245	Could not resolve this reference. Could not locate the assembly "SolidWorks.Interop.swpublished". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	TestRunner	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets	2110	
Warning	MSB3245	Could not resolve this reference. Could not locate the assembly "SolidWorksTools". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors.	TestRunner	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets	2110	
</details>

brawner, 2020-10-12:
> but is still not accessible under Tools

The "Export to URDF" is available under the File menu. Putting it in tools or creating a toolbar might be a good feature for the future.

http://wiki.ros.org/sw_urdf_exporter/Tutorials/Export%20an%20Assembly

mis-eu, 2020-10-12:
> _question:_
>
> do these Reg keys play a role in API compatibility with particular SW release?
>
> ```
> [Run]
> Filename: [...]
```

acbuynak, 2020-10-12:
Using the above provided (unsigned) .exe file.. https://github.com/ros/solidworks_urdf_exporter/issues/105#issuecomment-705105298

I was successfully able to add the SW2URDF tool and utilize the functionality (assign joints/links)
I have not yet confirmed the final export of the URDF from a sldprt/sldasm.

mis-eu, 2020-10-13:
> The "Export to URDF" is available under the File menu.

on  Windows 18363, SolidWorks SolidWorks Standard, the updated executable installer [sw2urdfSetup.exe](https://github.com/ros/solidworks_urdf_exporter/blob/58c2f2a7c55a5084618ef1c3007defb5a2b9ef51/INSTALL/OUTPUT/sw2urdfSetup.exe?raw=true) does not seem to work (no URDF Exporter under FIle):

[image]

brawner, 2020-10-16:
This was addressed in #106 and has now been released as version 1.6! https://github.com/ros/solidworks_urdf_exporter/releases/tag/1.6.0

closed 2020-10-16 by brawner

srsidd, 2021-01-07:
I'm having trouble with V1.6 with Solidworks 2020.. I the the `Export to URDF` option under tools. However when I click that, nothing happens, I see a window open and close within a fraction of a second. Anyway to debug this?