ros/solidworks_urdf_exporter #107: solidworks_urdf_exporter is disabled in solidworks 2020

jaytjoa, 2020-10-08:
Hi, I am using SolidWorks 2020 - Student Edition. I used the add-ins before in the student version in the SLDWRKS 2019. When I upgraded it and reinstalled the add-ins. The plug in was not recognized at all. Trying also to install it with source codes but some errors are shown.

Anybody with this experience?

brawner, 2020-10-08:
Please see #106 and try the new installer I linked below. Let me know if that works for you so I can see if it fixed the issue for everyone.

jaytjoa, 2020-10-08:
Hi, the issue in the SOLIDWORKS 2020 is solved by using the exe in #105. Thank you for good support!

brawner, 2020-10-08:
I need people to review and test it out that installer. Please try it out and comment on #105 if it works. This issue is a duplicate and will be closed.

closed 2020-10-08 by brawner