ros/solidworks_urdf_exporter #110: Create the URDF as .urdf.xacro, instead of .urdf

AndyZe, 2020-12-28:
A very minor suggestion: it would be nice to save the URDF as ".urdf.xacro" instead of ".urdf" so we can automatically start using substitutions like `${pi}`.

I can make a PR if you want me to. I guess it should be a very easy fix.

AndyZe, 2020-12-28:
Basically I think what would need to happen is:

- Rename the file with the `.urdf.xacro` extension, instead of `.urdf`

- Make this replacement in the launch files:

```
  <param
    name="robot_description"
    textfile="$(find arm_pkg)/urdf/arm_name.urdf,xacro" />
```

should become:

`<param name="robot_description" command="$(find xacro)/xacro $(find arm_pkg)/urdf/arm_name.urdf.xacro" />`

brawner, 2020-12-28:
Creating a xacro compatible file has been a long-desired feature of this exporter and is in the works for version 2.0. Though, I think it would take a bit more development than the changes you are proposing.

AndyZe, 2020-12-28:
Well, I was able to start using xacro with those same changes this morning. I think it's the minimum to get xacro working

brawner, 2020-12-28:
> I think it's the minimum to get xacro working

The main sticking point (to me) is that xacro should be a feature that users can opt into. Forcing everyone to use xacro is undesirable, especially for the non-ros users, and therefore adding a xacro feature would require some UI redesign. Secondly, there is more that I would want to see in a full `xacro exporter` than simple constant replacements.

But don't worry, this feature will be integrated into V2, and I would be happy with your feedback once that's in suitable beta form

AndyZe, 2020-12-28:
Sounds good, feel free to close if you want

brawner, 2020-12-28:
I used to have an open issue for xacro on bitbucket, but I never transferred it over. I'll leave this one open to catch for more discussion if people have thoughts.