ros/solidworks_urdf_exporter #111: When converting the assembly, the axis and coordinate system of the part cannot be read

ycpss91255, 2021-01-21:
Hello there!
When converting to URDF, the axis and coordinates of the part cannot system be read
[image]

brawner, 2021-01-21:
Hmm, strange indeed. Do you have a copy of a model that demonstrates this issue?

ycpss91255, 2021-01-22:
[全向輪.zip](https://github.com/ros/solidworks_urdf_exporter/files/5853062/default.zip)
This is part of my model.
This problem can be reproduced by opening OWMR.SLDASM.
After testing, I found that it may be an assembly, and this error occurs when opening another assembly.

brawner, 2021-01-22:
I think I realized what the issue here is. To prevent the tool from searching all the way down the tree of assemblies, it only looks for reference geometry in the top level and immediate children of the top level assembly. So it won't look more than one level down.

Try creating the geometries at the top level that are linked to the geometries at the lower levels. Do those show up?

ycpss91255, 2021-01-22:
Yes, he will appear.
So if I need to let him search down, is there a way to adjust it?

brawner, 2021-01-22:
Unfortunately, this is not possible currently.

The recommended solution is to organize the model such that there is one assembly per link. But looking at the names of your model, you have each roller in an omni wheel as a separate link? That might be overkill unless you need that sort of fidelity for simulation.

If you do need the whole omni wheel with separate links for each roller, then you might try exporting just the wheel as a urdf file. Edit that file to be a xacro macro (http://wiki.ros.org/xacro), then incorporate that back into the exported main assembly. 

I think this issue can be closed though. If you have further questions, please post them to answers.ros.org.

closed 2021-01-22 by brawner

ycpss91255, 2021-01-23:
thank you ,I will try to use xacro

divserge, 2023-05-03:
Hi! Could you please point me to a place in the code, where this top-level cut-off happens? I'd like to improve the tool, by optionally allowing it to search deeper.

divserge, 2023-05-03:
@brawner

brawner, 2023-05-03:
For context, this greatly impacts large assembiles. This used to be the case, but was reduced to top level assemblies (#17). Please make a proposal about what this feature would look like and how it would be implemented. The amount of work to do this well is not trivial