ros/solidworks_urdf_exporter #112: URDF add-in exits/crashs suddenly and leaves nothing

Undefinedefity, 2021-02-06:
I installed the sw_urdf_exporter v1.60 and every time I try to use it, say click "Tools>Export to URDF", then the new poped window flashes and exits/crashes suddenly, but the Solidworks itself stays there. This problem happens in both my SolidWorks 2018 and SolidWorks 2019 (I have installed both at the same time on my PC).

Undefinedefity, 2021-02-06:
I just found that if I changed it into the older version, say v1.5.1, it works.

brawner, 2021-02-06:
Yah, 1.6 was updated for SW 2020 and may not be compatible with previous versions.

FrancisGS131, 2021-02-15:
Hi, I am using the sw_urdf_exporter v1.60 in SolidWorks 2020. When I click "Tools>Export to URDF", the pop-up window also immediately flashes and exits/crashes. SolidWorks window remains but I can't get the pop-up window to open. Is this an issue that others are also encountering?

FirstCaptain, 2021-05-03:
Getting the same error in that same version @FrancisGS131  any update or work arounds?

FrancisGS131, 2021-05-04:
I couldn't figure out a workaround with the 2020 version. I had access to a 2019 version at the time and used that instead @FirstCaptain It doesn't seem like this solidworks-urdf exporter is working for the latest version of SolidWorks

Rapha-z, 2021-08-24:
I can confirm this issue (SW 2020 | sw_urdf_exporter v1.60). 

> Hi, I am using the sw_urdf_exporter v1.60 in SolidWorks 2020. When I click "Tools>Export to URDF", the pop-up window also immediately flashes and exits/crashes. SolidWorks window remains but I can't get the pop-up window to open. Is this an issue that others are also encountering?

zythhh, 2021-09-10:
> 嗨， 我在固体工厂 2020 中使用sw_urdf_exporter v1.60 。当我单击"工具>出口到 URDF"时，弹出窗口也会立即闪烁和退出/崩溃。固体工厂窗口仍然存在， 但我不能让弹出窗口打开。这是别人也遇到的问题吗？

你好，请问你现在有没有找到什么有效的解决方法，如果找到了能不能告知一下，万分感谢

brawner, 2021-11-21:
A new installer has been released for 2021 and v1.6.1. The code itself seems continue to work, but it seems the packaging of it can be fragile.

I'm going to suggest from now on, that anyone who has trouble with the prepackaged installer, to first try installing it by source. See https://wiki.ros.org/sw_urdf_exporte for instructions.

closed 2021-11-21 by brawner