ros/solidworks_urdf_exporter #113: "Reference sketch does not exist"

wh366666648, 2021-02-28:
Hi,

I was encountering a problem with a configuration property manager that the *" reference sketch does not exist"* during SOLIDWORKS exported a model to urdf.

This happened when I tried to add several reference geometries such as coordinate systems for export preparation to urdf.

Though this would not occur when I export directly with the axes and coordinate systems left as automating generated, the model would not perform correctly in this case, signifying the necessity of defining these by myself.

Therefore the previous question comes again.

Could anyone provide any help regarding this problem? (I did not find related problems in the tutorials. Thus I hope someone might help with this issue.)

Thank you for all this.

wh366666648, 2021-02-28:
So the problem may be reconsidered as `in what condition the reference geometry is invalid` during the export, I think. How to avoid this and create valid reference geometries?

marv-le, 2021-05-23:
Hey @wh366666648,

I had the same problem, after I deleted the URDF Export Configuration, the URDF Refernce and the Origins and Axis for the Joints from the assembly. Unfortunately i found no smart solution. So what has worked for me was to create a new assembly for my robot from scratch again and then do the URDF export again. It's a bit unconvenient but it works.

wh366666648, 2021-05-23:
Thanks @marv-le 
Actually, I found that this problem DISAPPEARS when I tried to export again from an unconfigured assembly. The procedures were almost the same. Therefore I think perhaps that would be a bug of the exporter (?)
And I now believe that saving the assembly before the file is added with reference geometries might be useful and necessary.

Thank you for replying regarding this issue. Best regards.

xin-Dream, 2021-11-23:
你好，我也遇到了这个问题。
我想我应该是解决了。在为link选择转轴和坐标系时，不要使用自动生成的坐标系。对于固定的配件可以选择使用base_link。这样在导出的时候就不会出错了。

Hello, I have the same problem.
I think I have it figured out. Do not use an automatically generated coordinate system when selecting the rotation axis and coordinate system for Link. You can choose to use base_link for fixed accessories. So that you don't make mistakes when you export.

Bassem-Touati, 2023-04-09:
> Thanks @marv-le Actually, I found that this problem DISAPPEARS when I tried to export again from an unconfigured assembly. The procedures were almost the same. Therefore I think perhaps that would [...]

i don't understand please can you exlplain how to resolve this problem

patrik1994lindstrom, 2023-04-18:
I had created a global coordinate system that I wanted to use, that caused this issue. By not using it I was able to use the local coordinate systems and axis

luoye2333, 2024-01-05:
i set automatically generate in the end-effector of the robot (which is a fixed joint)
give that link a coordinate system and axis and not use automatically generate solves my problem

CYTMWIA, 2024-04-08:
Delete URDF Reference and axis solved my problem.
[image]

dani71153, 2024-10-25:
"
I had created a global coordinate system that I wanted to use, that caused this issue. By not using it I was able to use the local coordinate systems and axis

"

This solved the problem for me (I deleted that configuration). So likely if you found a video on youtube that uses the creation of a  global coordinate systems, the best way to solve this is assembling your robot with the ROS convention of XYZ.

More for info here:

https://www.ros.org/reps/rep-0103.html

zid-tah, 2024-12-28:
hi 
is this error related to compatibility issues , because i have solidworks 2024 and urdf exporter 2021 and i have the same error

dani71153, 2025-07-28:
I'm coming back to this because I updated my robot. The definitive solution is to NOT USE ANYTHING AUTOMATICALLY GENERATED BY THE PLUGIN. You’ll have to define the reference axes and their orientations yourself.

I ran into this error again. It’s definitely something related to cache or a stored configuration, because the first time I tested this new version of my robot, it worked — though I had other issues due to incorrect configurations. After that, it never again accepted the global reference I designed, or applied it correctly to the wheels.

I did manage to include the reference manually, and that’s when — if you look at the child links — the base_link behaves correctly with global coordinates, and with the reference frame you created. But if you go into a child link and leave its frame as automatically generated, it fails.

So the problem is that auto-generated frames on child links do not respect or inherit the manually defined global reference.