ros/solidworks_urdf_exporter #114: Output mesh format

XDinEuro, 2021-03-03:
STL as a mesh format, only contains vertices information of a mesh, for better visualization, can we have .OBJ as output format, for keeping material. texture, etc information, for better visualization?

brawner, 2021-11-21:
It doesn't look like SolidWorks supports .OBJ exports by default, but it could be done with an add-in? Probably won't happen until I build out v2 of this tool.

http://help.solidworks.com/2021/english/SolidWorks/scanto3d/t_exporing_files.htm

akjay, 2022-10-14:
Does STEP file supportted by SolidWorks?I want to output a STEP file since it can keep color information for better visualization.

brawner, 2022-10-15:
Step is recognized by Solidworks, but is not generally recognized by any tool that consumes URDFs.

akjay, 2022-10-15:
> Step is recognized by Solidworks, but is not generally recognized by any tool that consumes URDFs.

Actually I build a softeare myself which use URDF format to show a robot，and use STEP files for a better presentation that STL since stl not contains any color information，so I wonder this plugin can output STEP files or not，maybe I can fork one and make the chnage myself to achieve this goal，but not know how to use SolidWorks SDK API，is it easy to do？

akjay, 2022-10-18:
Just tried that change the file's extension name to STEP will generate a STEP file with function "SaveAs", problem solved.Thank you for this excellent project. @brawner

ycpss91255, 2023-06-07:
> Just tried that change the file's extension name to STEP will generate a STEP file with function "SaveAs", problem solved.Thank you for this excellent project. @brawner

Hi, I have the same problem, can you share your solution?

KenMat765, 2023-12-05:
After changing the output to 3dxml, I was able to reflect the colors.

[image]