ros/solidworks_urdf_exporter #117: Wrong inertia values in the exported URDF

kakarot9996, 2021-05-22:
When defining links made up of multiple parts in an assembly, the respective inertia values at the centre of mass of the individual parts (after using similarity transformation to make it parallel to the joint frame) is directly added to give the net inertia matrix instead of first translating to the net centre of mass (of all parts in the link combined) and then adding. This gives the wrong inertia matrix. This issue was only spotted recently, kindly resolve it at the earliest.

jespersmith, 2021-07-29:
I noticed this issue years ago and left a issue report on the bitbucket. Unfortunately, I cannot pull this up anymore but this is problematic for force controlled robots.

dsubhasish09, 2021-07-29:
Yes, we are doing the same but it's a pain to manually alter the inertia values for every link, especially if you have a large number of these. Are you automating this by any means?

jespersmith, 2021-07-29:
We have a simple UI to edit URDF's, that can parse the data from the solidworks mass tool (copy to clipboard data) . We just picked up development on it again, and we might consider open sourcing it.

brawner, 2021-11-21:
That is unfortunate to hear about this bug, as I pretty much just use SW's interfaces to calculate the MOI. It sounds like it may be a bug upstream.

https://github.com/ros/solidworks_urdf_exporter/blob/master/SW2URDF/URDFExport/ExportHelperExtension.cs#L248-L260

Do either of you have representative models and a URDF to help make comparisons?

bbokser, 2022-05-03:
I can provide representative models, if you would like. I would much appreciate not having to manually correct the values every time I generate a URDF. Goodness knows how many headaches this has caused over the years...

xwavex, 2022-08-04:
This sound like something that urgently needs to be fixed

qkevinma, 2023-01-31:
I'd like to add fuel to this fire. The inertia calculation is incorrect and needs to be fixed ASAP. Our workaround is to manually enter the correct inertia matrix values with every link that has more than two parts or bodies. The exporter only works properly if each link has a single body. Even multiple bodies within the same part will still throw off the inertia calculation.

brawner, 2023-01-31:
This code is open source, PRs are always welcome

luoye2333, 2024-06-03:
> We have a simple UI to edit URDF's, that can parse the data from the solidworks mass tool (copy to clipboard data) . We just picked up development on it again, and we might consider open sourcing it.

I use chatgpt to generate a convert program in python. When use, copy the text in solidworks mass tool to the const string in the code. Then copy the output text to the original urdf file. Feel free to use.

```python
# 原始字符串
raw_string = """
所选零部件 的质量属性
     坐标系： BCS

质量 = 5439.19 克

体积 = 2525851.14 立方毫米

表面积 = 979087.02  平方毫米

重心 : ( 毫米 )
	X = 0.00
	Y = 1.10
	Z = 7.33

惯性主轴和惯性主力矩: ( 克 *  平方毫米 )
由重心决定。
	 Ix = ( 0.00, -1.00, -0.01)   	Px = 45020646.47
	 Iy = ( 1.00,  0.00,  0.00)   	Py = 48410923.95
	 Iz = ( 0.00, -0.01,  1.00)   	Pz = 80241299.66

惯性张量: ( 克 *  平方毫米 )
由重心决定，并且对齐输出的坐标系。 （使用正张量记数法。）
	Lxx = 48410923.95	Lxy = 1.36	Lxz = -0.97
	Lyx = 1.36	Lyy = 45021610.36	Lyz = 184249.38
	Lzx = -0.97	Lzy = 184249.38	Lzz = 80240335.77

惯性张量: ( 克 *  平方毫米 )
由输出座标系决定。 （使用正张量记数法。）
	Ixx = 48709740.16	Ixy = 0.92	Ixz = -3.91
	Iyx = 0.92	Iyy = 45313794.64	Iyz = 228269.22
	Izx = -3.91	Izy = 228269.22	Izz = 80246967.70
"""

# 提取数值并除以10^9
mass = str(float(raw_string.split('质量 = ')[1].split()[0]) / 1e3)
cog_x = str(float(raw_string.split('X = ')[1].split()[0]) / 1e3)
cog_y = str(float(raw_string.split('Y = ')[1].split()[0]) / 1e3)
cog_z = str(float(raw_string.split('Z = ')[1].split()[0]) / 1e3)
ixx = str(float(raw_string.split('Ixx = ')[1].split()[0]) / 1e9)
ixy = str(float(raw_string.split('Ixy = ')[1].split()[0]) / 1e9)
ixz = str(float(raw_string.split('Ixz = ')[1].split()[0]) / 1e9)
iyy = str(float(raw_string.split('Iyy = ')[1].split()[0]) / 1e9)
iyz = str(float(raw_string.split('Iyz = ')[1].split()[0]) / 1e9)
izz = str(float(raw_string.split('Izz = ')[1].split()[0]) / 1e9)

# 输出格式
output_string = f"""
<inertial>
  <origin
    xyz="{cog_x} {cog_y} {cog_z}"
    rpy="0 0 0" />
  <mass
    value="{mass}" />
  <inertia
    ixx="{ixx}"
    ixy="{ixy}"
    ixz="{ixz}"
    iyy="{iyy}"
    iyz="{iyz}"
    izz="{izz}" />
</inertial>
"""

print(output_string)
```

jespersmith, 2024-06-19:
> I use chatgpt to generate a convert program in python. When use, copy the text in solidworks mass tool to the const string in the code. Then copy the output text to the original urdf file. Feel free [...]

Note that the off-diagonal terms need to be negated between Solidworks and URDF (inertia matrix vs tensor as far as I remember)

From my personal docs
```
WARNING: SolidWorks shows the moments of inertia, URDF expects the inertia tensor matrix.

The inertia tensor matrix is defined below from the moments of inertia:

![afbeelding](https://github.com/ros/solidworks_urdf_exporter/assets/8227626/70b210a0-026c-4544-9933-1491e382c23c)

Pretty much, make the off-diagonal values negative when exporting.
```

luoye2333, 2024-06-21:
Thank you [jespersmith](https://github.com/jespersmith). You are right. I have edited the code https://github.com/ros/solidworks_urdf_exporter/issues/117#issuecomment-2144241339 and added the minus symbol '-' in ixy ixz and iyz .

wenhao12111, 2024-11-16:
mark +1

ZeroL223, 2025-04-16:
Change the inertia tensor (Crossproduct Convention) to Negative Tensor Notation in SolidWorks, so that off-diagonal terms don't need to be negated

cedrichld, 2025-07-12:
So is it enough to "change the inertia tensor (Crossproduct Convention) to Negative Tensor Notation in SolidWorks, so that off-diagonal terms don't need to be negated" just like @ZeroL223 said? sw_urdf_exporter will work without issues?

bbokser, 2025-07-13:
@cedrichld No, that is a separate problem that should probably have its own Issue. The problem described by OP is that the software fails to apply the parallel axis theorem correctly. The inertia tensors that come out of this software will continue to be incorrect until _both_ bugs are fixed.