ros/solidworks_urdf_exporter #124: [help me]SW2021,install ok,but not normal work

zouzhe1, 2022-04-13:
* win10 x64
* NET4.8
* SW2021
1,install URDF tool success
2,can see this plug-in in SW2021
3,when click the tool,new window flash by
4,No files generated

please，WHY？

brawner, 2022-04-13:
Can you confirm which release you downloaded?

What level of solidworks are you using?

zouzhe1, 2022-04-13:
@brawner   my down link    https://github.com/ros/solidworks_urdf_exporter/releases/download/1.6.1/sw2urdfSetup.exe
，and  i use SW2021 premium (win10 professional x64 21H1 2121/03/11)，thanks.

brawner, 2022-04-13:
That setup has worked for me on my machine. To help diagnose the issue, can you install this project via source. Running with MS visual studio in debug mode can help point to where the crash is happening. Also, this plugin creates a log file in your home directory. It would help to upload that as well

zouzhe1, 2022-04-19:
@brawner  can you share a SW api  2021 file? i registor the sw web fault....

zapplelove, 2022-09-02:
https://user-images.githubusercontent.com/18183148/188118900-f0530b57-df30-4870-b582-96fef9fd0ebd.mp4

ycpss91255, 2022-09-28:
Me too, my SolidWorks is 2022 verison

brawner, 2022-12-06:
The SW API file is provided with your SolidWorks installation, you have to install it separately.

https://help.solidworks.com/2021/english/api/sldworksapiprogguide/gettingstarted/solidworks_api_getting_started_overview.htm?verRedirect=1