ros/solidworks_urdf_exporter #127: link name: b'not available'

Senwei-Huang, 2022-06-08:
**System Information**
1. SolidWorks version and service pack
```
SOLIDWORKS 2021 SP0.0
win10 x64
```
3. Version of exporter
`v1.6.1 (SolidWorks 2021)`

**Describe the bug**
I used exporter to generate a URDF file, and use pybullet(a dynamic simulation software) to read the URDF file. 
Everything else is normal except that the names of each link in the URDF file cannot be read. When you try to get the name of the link, it returns b'not available'
```
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
link name: b'not available'
```
**Attach an example of a corrected URDF (if applicable)**
One of the links in the URDF file
```
</link>
  <link name="FSpine_Link">
    <inertial>
      <origin xyz="0.21853 8.8784E-06 -0.055766" rpy="0 0 0" />
      <mass value="20" />
      <inertia
        ixx="4.2876"
        ixy="1.496E-05"
        ixz="2.6509"
        iyy="5.5072"
        iyz="-2.0325E-05"
        izz="4.0165" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://BigCat/meshes/FSpine_Link.STL" />
      </geometry>
      <material name="red">
        <color rgba="0.89804 0.91765 0.92941 1" />
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <mesh filename="package://BigCat/meshes/FSpine_Link.STL" />
      </geometry>
    </collision>
  </link>
```
I hope I can get your help. Thank you very much.
@brawner

brawner, 2022-11-04:
Do you have an example of the soldiworks model that created this issue and the URDF was generated?

brawner, 2022-12-06:
Without more information, I won't be able to reproduce this issue.

closed 2022-12-06 by brawner

Senwei-Huang, 2023-03-23:
> Do you have an example of the soldiworks model that created this issue and the URDF was generated?

Thank you, Brawner. I have solved this problem. This is caused by setting the parameter useMaximalCoordinates to True when I import a model using the loadURDF function in PyBullet. Thank you again for your reply. 
@brawner