ros/solidworks_urdf_exporter #130: URDF export error: parts mesh are in wrong positions in explo

Dream-build, 2022-09-08:
**System Information**
1. SolidWorks version and service pack:   SW2020 SP0
2. Version of exporter
v1.6.0 
**Describe the bug**
A clear and concise description of what the bug is.
I was exporting my robot from SW to urdf, at first time it went well, however the axis position of one of the joints was wrong. So I generate the urdf again, this time all axis was fine from the looking of SW. However when I opened the urdf file in the moveit! it shows that the model was not right, and parts exported are in wrong posistions.
This is the robot looks like in SW:
[image]
This is urdf opened in moveit:
[image]
**Attach simple model which clearly illustrates the issue**
[warp_wraith.zip](https://github.com/ros/solidworks_urdf_exporter/files/9524401/warp_wraith.zip)
**Attach an example of a corrected URDF (if applicable)**
Sorry, it was deleted, I didn't expect such would happen.

DexterxLab, 2026-08-21:
did you resolve the issue?
i am facing similar issue, my robot has all good orientations and links in SW but when exported and displayed in rviz its mostly wrong orientation. how to resolve it? made the urdf file many time already

brawner, 2026-08-24:
Does this PR fix your issue? https://github.com/ros/solidworks_urdf_exporter/pull/182