ros/solidworks_urdf_exporter #135: SW2URDF: Inferring the joint geometry failed...

dlerner97, 2023-02-06:
1. SolidWorks 2022
2. 1.6.0-4-g7f85cfe

Whenever I export an URDF, I fully constrain all SW2URDF parameters and do not rely on the automatic joint generation. Yet, very often I receive an error when I try to export them (as shown in the image below). This happens quite frequently when I have a single mesh/part but need to make a fixed joint between links at two locations on the part. I know this is an odd use case but it allows us to compose a sequence of URDF's at runtime by creating 6*0 joints between the output link of one part and an input link of a connected URDF. Because this is a fixed and fully constrained joint offset, this should be easy with the urdf exporter but very commonly errors out upon exporting. This also happens every so often when exporting actual assemblies with fully constrained mates as fixed joints. I cannot seem to find much of a pattern there but I'm hoping that fixing one will fix the other. I have attached an error message as well as a pack-and-go zip folder containing an assembly with the specific issue. I have also attached the logs in case they help.

[image]

[ infer_joint_geom_ex_assem.zip](https://github.com/ros/solidworks_urdf_exporter/files/10668973/infer_joint_geom_ex_assem.zip)
[sw2urdf.log](https://github.com/ros/solidworks_urdf_exporter/files/10669003/sw2urdf.log)

SammyRamone, 2023-04-25:
I had the same error message in my assembly. I just had two parts that I wanted to connect with a fixed link. Each of the parts had a sketch with the point marked where they are to be joined together. The first part was set fixed and the second one was connected by connecting those two sketeches without any DoF left. Manually setting the joint type at export to "fixed" did not resolve the issue.
I finally fixed this by also manually creating the joint origin and axis (by adding an coordinate system and axis at the point where both parts are connected). This resolved the issue and the export now works.

dlerner97, 2023-04-26:
Thanks for the input! The thing is, I always add my own coordinate systems and a fixed joint doesn't need an axis. Other than the axis, I never leave any input blank and input my own values for all of them and I still often get this error.

Carpetbagleader, 2025-02-14:
Is there any way to work around this a bug?

dlerner97, 2025-02-14:
There are a few ways, none of them are good or work every time. Usually, you can get by by suppressing one of the mates. The issue is that sw2urdf was built for robotic arms where fixed joints don't really exist. If you suppress a joint to make it seem prismatic or revolute, then it'll work. You'll just have to go on and post process the results. Things get a little trickier if you're a) just trying to define a point in space (i.e just a coordinate system) rather than a rigid body or b) if a component is fixed without any mates (for example if you're using a skeletal sketch to define the locations of each component). Then, you may need to a) add a dummy component, mate it to the assembly, and use the same cs. In the case of b) you'll need to completely remake the assembly with mates only.

All of this stems from one small bug: sw2urdf always tries to infer the joint type of a fixed joint even when you explicitly set it.

Hope this helps! Good luck!

thesamrooney, 2025-07-18:
Can confirm that this is still an issue. I found a workaround to switch all my fixed joints to prismatic joints. I label all my real prismatic/revolute joints with 'dof', i.e. dof_q1, dof_q2, etc, and the fixed joints with 'tf', like tf_base_frame, tf_j6_tcp, etc. I think it makes postprocessing a bit easier - hope it helps the next lost soul who finds this!