ros/solidworks_urdf_exporter #136: Export to URDF Window instantly closes

titus123abc123, 2023-02-21:
**System Information**
1. SolidWorks version and service pack - 2021-2022 (the logo has 2021 on it but it is called 2021-2022 for the academic year) 
2. Version of exporter - latest release 1.6.1

**Describe the bug**
A clear and concise description of what the bug is.

Clicking Export URDF in Tools opens a window for a split second and then instantly closes.

**Attach simple model which clearly illustrates the issue**

Any model causes this issue

**Attach an example of a corrected URDF (if applicable)**

No URDF because exporter won't open

**Potential Lead**

I also have SOLDIWORKS2022 installed on my system. I started by installing the latest version of the exporter, praying it would work on SW2022. It didn't so I installed SW2021 and it has the exact same problem.

Please let me know if there are logs or anything I can attach.

Thank you!

brawner, 2023-02-21:
Based on feedback in https://github.com/ros/solidworks_urdf_exporter/issues/134, it is possible the installer is too out of date. For now, until I have time to run another release, you may have to compile and install by source.

bibinthomas007, 2023-03-03:
I tried to compile and install by source by following the development tutorial in here.
https://github.com/ros/solidworks_urdf_exporter
[image]

These are the error codes i got 
CS0246
CS0538
CS0616

brawner, 2023-03-03:
You need to install the SolidWorks API tools manually from your installer