ros/solidworks_urdf_exporter #145: Has this plugin stopped updating？

wxydejoy, 2023-08-18:
Has this plugin stopped updating？
@brawner @mcevoyandy @davetcoleman @Extent421 @tonybaltovski @sachinkum0009 @708yamaguchi 

Please don't give up

brawner, 2023-09-19:
Likely the binary just needs to be updated for the next release of SW. You can build it yourself according to the source directions on the wiki

wxydejoy, 2023-09-23:
maybe we need package for ros2 ?

gwitkowski2000, 2023-10-08:
@wxydejoy did you manage to get sw2urdf working with SolidWorks version 2022?

wxydejoy, 2023-10-11:
@gwitkowski2000 So far, the plug-in can work normally with SolidWorks2022, but the version of the generated software package is only ROS1. You also need to manually modify the package.xml and CMakelist.txt in the software package.

gwitkowski2000, 2023-10-17:
@wxydejoy it works indeed, perhaps i was looking in the wrong tab since it was in "Tools" instead of "File", where I was looking for it. I'm going to work with pyBullet, so unfortunately I don't have any experience with ROS2

closed 2024-04-01 by brawner