ros/solidworks_urdf_exporter #147: sw2urdf window appears and then quickly disappears

Femidis, 2023-11-06:
**System Information**
1. Solidworks Premium 2020 SP1
2. Windows 10 64x
3. Visual Studio 2022 Community

I tried to install using pre-compile version 1.16.0, but when i try Export to URDF, plugin window blinks without any effect.
I try to install using compile way, but have some problems with install SolidWorks API SDK, all links outdated. When i finally found link and try to compile, plugin appears in Add-Ins, but i can't activate it. F tick in table Activated always disappers, and button Export in URDF not appears regardless of the model.

SolidWorks API i found in NuGet in Visual Studio, for installing i used instruction from Github

shin0805, 2023-12-21:
When trying to use the URDF exporter with an sldprt file, I encountered a similar issue, but it didn't occur with an sldasm file : )

weianweigan, 2024-08-27:
I forked this repos and fixed it. There is a release in [forked repos](https://github.com/weianweigan/Solidworks_urdf_exporter).

brawner, 2024-08-27:
Would you be willing to make a PR?

brawner, 2024-08-27:
What change actually addressed this issue? Perhaps you could just make a PR from that?

weianweigan, 2024-08-27:
> What change actually addressed this issue? Perhaps you could just make a PR from that?

Ok

commit ee910b25 (2024-08-27): Fixed #147 (#158)
closed 2024-08-27 by brawner