ros/solidworks_urdf_exporter #162: Add Step-by-Step Guide for Exporting URDF Files from SolidWorks

sherif1152, 2024-10-02:
Hello,

**I working on the export process for URDF files from SolidWorks. I have written a detailed description and step-by-step guide that I believe will be helpful for users looking to perform this task. You can view the description here: [export_solid_as_urdf.md](https://github.com/arab-meet/3.URDF_Simulation_Workshop/blob/main/B.%20URDF/sw2urdf/export_solid_as_urdf.md).**

**I would like to create a pull request and add this description to the main branch of your repository. I believe it will greatly benefit users interested in the URDF export workflow from SolidWorks.**

**@brawner  @weianweigan**

Cheny5863, 2024-12-24:
very good job!

brawner, 2024-12-27:
It seems like a very good tutorial for creating your robot is SW and exporting with the tool. But there is already an in-depth tutorial over at https://wiki.ros.org/sw_urdf_exporter. What changes do you think need to be made to those tutorials?