ros/solidworks_urdf_exporter #165: update to ros2

QILIANGDELEISHUI, 2024-11-22:
I am a beginner of ros2, thank you very much for developing this plug-in, which helps me to build complex models in visual graphics software and can be easily displayed. I recently encountered an issue where urdf files generated using this plugin and STL files in the meshes folder could no longer be visualized in gazebo, in the ros2 environment. I can't solve the problem, and I can't judge the source of the problem. However, in 2024, ros1 is already lagging behind, would you consider upgrading this plugin to ros2? . Requests from beginners. Thank you very much!

fish1sheep, 2024-12-21:
I created a small tool in my repository that can convert the directory generated by sw2urdf into a format compatible with ROS2. It allows the model to be displayed in both RViz2 and Gazebo. I have already tested it and confirmed it works on Humble.

zahra-ouirghimmi, 2025-06-29:
> I created a small tool in my repository that can convert the directory generated by sw2urdf into a format compatible with ROS2. It allows the model to be displayed in both RViz2 and Gazebo. I have already tested it and confirmed it works on Humble. https://github.com/fish1sheep/sw2urdf_ROS2.git

can u help me please! i used your link and i get this error 
Error: SDF parsing the xml failed.
Traceback (most recent call last):
  File "/home/user/Downloads/sw2urdf_ROS2-master/dir_ros2.py", line 255, in <module>
    subprocess.run(command, shell=True, check=True)
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'ign sdf -p /home/user/Downloads/sw2urdf_ROS2-master/arm_description/urdf/arm_description.urdf > /home/user/Downloads/sw2urdf_ROS2-master/arm_description/urdf/model.sdf' returned non-zero exit status 255.

Zahra149, 2025-07-01:
> I apologize, but this tool currently only supports the older version of Gazebo 11. I haven't adapted it for the newer version of Gazebo yet.

I understand, thank you for clarifying. Would it be possible for you to adapt it for me? I need it for my final year internship project.